Gabriel-LG/HA_Mower #1: Question

EmilEntchev, 2024-07-04:
The schematic has only the connections for the charging voltage, rain sensing and the accelerometer. Where are the connections for turning on the mower and turn off? Did you connect to the on button?

nsleigh, 2024-07-04:
Not the author but as I understand it the rain sensor is used to control running the mower.

Gabriel-LG, 2024-07-04:
@nsleigh you are correct, the rain sensor is actually used to control the mower.
@EmilEntchev It works as follows: The mower is set to mow 24 hours per day, except when it is raining. Now the ESP will "inject" a fake rain signal into the sensor, to keep the mower from starting. When the ESP receives the command to start mowing, it will release the rain sensor so the robot starts mowing. When the time is up, the ESP asserts the rain signal again, making the robot return to its dock.

nsleigh, 2024-07-04:
@Gabriel-LG thanks, trying to find a similar way to control my Flymo Easilife but that mows in the rain!

closed 2024-07-04 by EmilEntchev