Gabriel-LG/HA_Mower #4: The ESP battery drain makes the Parkside PMRA 20-Li-A1 refuse for some periods

Gabriel-LG, 2024-07-08:
When the battery is full, the charger floats the battery.
When the battery drops below 18.9V, the charger tops the battery off.
18.9V corresponds with approximately 60% battery level. The mower refuses to start if the battery level is below 70%.
This causes the mower to refuse to mow for a period of about 4 hours per day.

Possible solution:
Add a deep sleep, where the mower status is updated less frequently (e.g. once every 5 minutes). Hopefully, this will cause less drain on the battery, enough to make it to the next periodical recharge (if any).

[image]

Gabriel-LG, 2024-07-11:
I implemented a quick and dirty rate limiting during standby (send battery and accelerometer once per 5 minutes).
Also enabled power saving at high.
The battery drain is exactly the same, except there is more variance on the battery voltage level.

Gabriel-LG, 2024-07-13:
I created a firmware that does only pulls down the rainsensor and report the battery voltage ever 5 minutes.
This extended the time between recharging to about 23:30 hours.
- This may help https://www.reddit.com/r/Esphome/comments/19ew3lv/state_of_unused_pins_in_esphome
- Also increasing the resistor tapping into the rainsensor will save some power.
- I am curious as to how much the mower itself consumes. Removing the battery for a day may shed some light on that.

nsleigh, 2024-08-10:
> I implemented a quick and dirty rate limiting during standby (send battery and accelerometer once per 5 minutes). Also enabled power saving at high. The battery drain is exactly the same, except there is more variance on the battery voltage level.

Is this in github? Mine drains but I think it might do that anyway as if you start it (without sensor) it charges for a while before starting.

quenthal, 2025-05-10:
> I created a firmware that does only pulls down the rainsensor and report the battery voltage ever 5 minutes. This extended the time between recharging to about 23:30 hours.
>
> * This may help [...]

Could one use 10kohm or even up to 100kohm resistor tapping to the rainsensor?

I haven't yet started on my project (got Parkside PAMR 500 A1, which doesn't have app control, but still weekday calendar, timeslots available twice a week day and rainsensor, using the same Parkside batteries. 

I can naturally measure it myself, but is there some general guideline to recognize which part of the sensor needs to be pulled down (i.e. connected to GPIO13?

Gabriel-LG, 2025-05-10:
> Could one use 10kohm or even up to 100kohm resistor tapping to the rainsensor?
>
> I haven't yet started on my project (got Parkside PAMR 500 A1, which doesn't have app control, but still weekday [...]

If the rainsensor is similar to the PMRA20, then there will be 2.7V potential over the contact points of the rain sensor (assuming it is dry). Connect the ESP pin to the positive side.
The resistor is needed, to protect the ESP GPIO pin from over current. At 10kOhm I don't expect it to work. There will only be current flowing thought the resistor while mowing, so it will not drain the battery.

RELA20, 2025-05-28:
Hi, I want to add this to my pamrc 250 mower (also parkside). It has app control but only over Bluetooth. The rain sensor on this model consists of two prongs that protrude trough the housing of the mower. The voltage measured between the prongs is 3,2V when dry. When I switch GPIO13, the voltage drops to 0,58V. I have used a 270ohm resistor between GPIO13 and the positive side of the rain sensor. When I press the switch in home assistant, the state of the mower on home assistant never goes to on.

The rest of the sensors seem to work as expected. Only starting (and I think stopping) the mower does not seem to work.

Do you have any suggestions what I could do? Maybe change resistor value? Or maybe I have setup the mower configuration wrong in the parkside app? (Could you share your app settings?)

Gabriel-LG, 2025-06-02:
> Hi, I want to add this to my pamrc 250 mower (also parkside). It has app control but only over Bluetooth. The rain sensor on this model consists of two prongs that protrude trough the housing of the mower. The voltage measured between the prongs is 3,2V when dry. When I switch GPIO13, the voltage drops to 0,58V. I have used a 270ohm resistor between GPIO13 and the positive side of the rain sensor. When I press the switch in home assistant, the state of the mower on home assistant never goes to on.
> 
> The rest of the sensors seem to work as expected. Only starting (and I think stopping) the mower does not seem to work.
> 
> Do you have any suggestions what I could do? Maybe change resistor value? Or maybe I have setup the mower configuration wrong in the parkside app? (Could you share your app settings?)
> 
> Edit: I managed to get it to work. The app was configured wrong.

Hi Rela20,
I create a new issue for your issue.
https://github.com/Gabriel-LG/HA_Mower/issues/8#issue-3111244645

Vasile-Hij, 2026-10-08:
I implemented this project on PAMR-500 robot back in May 2026 with 2A battery which died in July. Then I started with a battery of 4A and worked until this week  when in 5 days the battery died, despite it was in the docking station. Unfortunately, I had no voltage_battery sensor activated in HA, but I am thinking that we have a huge consumption at one point. Also I reckon to deactivate sending data when robot is out of any use (if in HA is setup to run 2 days a week, the other 5 days to be likely off or a wee ping of data at 8h). If you have any idea how to bring back to life the parkside batteries it would be nice.

vilhelmericson, 2026-10-08:
I also had the battery drain problem, it was solved like this;
a "dead" battery can be "revived" using another battery, see https://youtu.be/80iXbw9QDic?is=CJDAiFBpcsy2dGbf