Gabriel-LG/HA_Mower #6: Q: battery_voltage_gain

nsleigh, 2024-08-07:
First, thanks for a great project. I have a version built and am working on integrated to my Flymo mower.

How did you arrive at the _battery_voltage_gain_ value, is it just trial and error? (I used different resistors, just because they were all I have available 10k and 1k.)

Gabriel-LG, 2024-08-09:
Nice to hear that you have built a version also :)
The battery voltage gain was based on the resistor values,  but it was off by a few percent (due to electrical tolerances I guess), so I corrected the gain (assuming there is no offset error).

If your resisors are presice enough (and the ones on the D1 mini as well) then the formula to calculate the gain would be as follows:
**_(6.8kΩ+1kΩ) / 1kΩ * 3.3V/1.0V = 25.74_**
In your case this would be:
**_(10kΩ+1kΩ) / 1kΩ * 3.3V/1.0V = 36.3_**

The 3.3V/1.0V comes from the D1 mini's internal divider that scales the ADC input to 0 to 3.3V. If you are using a different board, then you may need to change this as well.

Also note that your battery voltage may not exceed 36.3V, as that will be out of spec for the ADC and could possibly damage the ESP (I actually don't know what absolute maximum rating is for the ESP8266's ADC).

nsleigh, 2024-08-10:
Thanks, that matches how I thought it would work - I found 34 worked for me (Flymo is a 20V battery).

It is working well to detect if it is running but I need to find out how to control it and 'know' it is mowing (I might have to change to an ESP32 and see if I can send Bluetooth). I have played around with adding the temperature but it looks like the internals get quite hot when mowing. When I can control it I was hoping to stop it on frosty days.

I added WIFI strength parameter and was surprised how good it was over my garden.