Gabriel-LG/HA_Mower #13: Not mowing after returning to charge

infernos90, 2025-07-05:
Helo. My robot not mowing after return to charge, only stand in standby. It doesn't start mowing even though the battery is fully charged, and I give it the command to mow.

infernos90, 2025-07-06:
Update: After returning stay in "rain sensor activated" and not starting again untill i opened the lid, press ok and command again to mow from ha.

[image]

Gabriel-LG, 2025-07-06:
My mower has a setting, how long to wait after the rainsensor is deactivated.
It could also be an electrical issue, that the  microcontroller is still slightly pulling the sensor down.

To test for the last case, what happens if you disconnect the microcontroller and use something metallic (I suggest a neodymium magnet) to fake the rainsensor? If you remove the magnet and the mower starts immediately, then it could be an electrical issue. If it keeps displaying rainsensor activated, it is likely a setting in you mower.

Gabriel-LG, 2025-07-06:
Oh and check the voltage on the rainsensor with a multimeter, to see if it is really deactivated.

vilhelmericson, 2025-07-14:
Hi, I finished my build the other day and am experiencing the same behavior as reported by infernos90, "Rain sensor activated". Sometimes I manage to start mowing via HA after opening the lid and pressing the OK button, but not always, most times something seems to hang.
My mower is however a Parkside PMRA 20-Li **B2**, i.e. not a PMRA 20-Li **A1** for which the design is intended and which Gabriel-LG has. There may be a difference between the models that causes our mowers to have this behavior.
I have verified my circuit board and its soldering to the best of my ability and am convinced that it is "electrically correct".

Regardless of our problem, I think Gabriel-LG's idea and design are absolutely brilliant.